wiredopposite/OGX-Mini #134: X360 controller vibration died after single usage of OGX-Mini

Coolawy, 2025-06-01:
Hello 

The issue is like in the topic. I've build OGX-Mini with hope of using its calibration possibities for all of my Xbox 360 controllers after sticks swap to hall sensor sticks. This function works as a charm because I was able to reach circularity around 1%. I am connecting it using Microsoft wireless adapter back to my X360 console.
Only yesterday after a week of using it I noticed that I have no vibration in controller. I used X360 app for controller checking, pc and mac as well, no vibration. So then I connected controller directly to console - no vibration as well. I took another controller - same story. Then I took another 2 I have, which I never connected through OGX-mini, connected them directly to the console first and they were working fine. So I connected one through OGX - vibration not working anymore. I connected it then directly to the console and... no vibration again. I couldn't believe it and I took my last controller and killed vibration in it as well.....

Did anyone had the same issue? Is there a way to hard reset X360 controller to bring it back to life?

LeGittee, 2025-06-04:
Hi, I did not get this issue, but it seems that OGX mini has killed the PIO ports of my Pico board. Before that I had the same issue of the PIO ports dying with another similar program called "HID remapper". It seems that those programs are somehow able to fry components.

I had a problem with a wired Xbox 360 gamepad, I replaced the bumper buttons which weren't functional, for this I had to de-solder the trigger potentiometers as well, and when I re-soldered them on, the triggers were not functional anymore. I tried to re-re-solder them again in different fashions, with jumper wires but it did not work, it seems the gamepad's microcontroller got damaged somehow. I don't know of a way to reset the micro controller chips.

Idk if it could be a fried smd component, what kind of controllers did you use ? were they all Wireless Microsoft gamepads ?

On the wired ones the rumble motors seem to be connected to power through a transistor + ceramic capacitor and a diode, I am not entirely sure if the wireless ones are similar.

You could use a multimeter to test them if you have one. Also, as you probably know, there are pcb scans of the Xbox 360 controllers on "Acidmods.com" to help you see how components are connected on the board

mastercheff, 2025-06-04:
RIP 360 rollers =/ . At least a good cause guys, I'm pretty sure next update is imminent

rebeloom, 2025-07-13:
Same scenario, Three different controllers that no longer have vibration working on any system.

yourself99, 2025-07-15:
Same on my side.

I used original Xbox360 Wireless controllers with the original Wireless Adapter on my PC only.

To debug tried directly sending rumble-commands via `xboxdrv` and `fftest` on Linux without any luck. Same setup works on controllers that weren't connected to OGX-mini before. So it seems to brick the controller and not the wireless receiver. My guess would be some safety function on the controller itself blocking rumble functionality as there was no initial rumbling when connecting that could lead to a direct hardware default. 

So maybe some Factory Reset of the controller firmware would do the job. Does anyone know a way to completely reset the controller? I didn't manage to find any way on my quick research.

xmgprojects, 2025-11-02:
I'm experiencing the same issue as well. Not sure what was the firmware version on my pico when I first connected the Xbox 360 controller as I bought it ready made. Version 1.0.0.a3 same issue. Haven't connected my second controller to test it. Im using a clone wireless receiver. Tested both controllers on pc with the same wireless receiver on pc and rumble does not work on the controller that has been connected to the ogx mini. I even tried to connect it back to my Xbox360 bit the vibration option is greyed out.

faithvoid, 2026-01-29:
Interesting. I wonder if that means OGX-Mini is touching a value in the controller's firmware that it's not supposed to, in that case? Hopefully it's as easy as just unflipping it. 

Someone smarter than me weigh in, [could this potentially be the offending code](https://github.com/wiredopposite/OGX-Mini/blob/master/Firmware/RP2040/src/USBHost/HostDriver/XInput/tuh_xinput/tuh_xinput_cmd.h)? There's a notice about a certain value disabling the rumble function, but it's seemingly never called.

```   //Sending 0x00, 0x00, 0x08, 0x00 will permanently disable rumble until you do this: ```

Full excerpt below:

```
    namespace Xbox360
    {
        static constexpr uint8_t RUMBLE[]   = { 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00 };
        static constexpr uint8_t LED[]      = { 0x01, 0x03, 0x00 };

        namespace Chatpad
        {
            static constexpr uint16_t CMD_KEEPALIVE_1 = 0x1F;
            static constexpr uint16_t CMD_KEEPALIVE_2 = 0x1E;
            static constexpr uint16_t CMD_LEDS_ON_KEYPRESS = 0x1B;
            static constexpr uint16_t CMD_CAPSLOCK_1 = 0x8;
            static constexpr uint16_t CMD_SQUARE_1 = 0x9;
            static constexpr uint16_t CMD_PEOPLE = 0xB;
            static constexpr uint16_t CMD_BACKLIGHT = 0xC;
            static constexpr uint16_t CMD_CAPSLOCK_2 = 0x11;
            static constexpr uint16_t CMD_SQUARE_2 = 0x12;
            static constexpr uint16_t CMD_SQUARE_CAPSLOCK = 0x13;
            static constexpr uint16_t CMD_CIRCLE = 0x14;
            static constexpr uint16_t CMD_CIRCLE_CAPSLOCK = 0x15;
            static constexpr uint16_t CMD_CIRCLE_SQUARE = 0x16;
            static constexpr uint16_t CMD_CIRCLE_SQUARE_CAPSLOCK = 0x17;

            static constexpr tusb_control_request_t INIT_1   = { .bmRequestType = 0x40, .bRequest = 0xA9, .wValue = 0xA30C, .wIndex = 0x4423, .wLength = 0 };
            static constexpr tusb_control_request_t INIT_2   = { .bmRequestType = 0x40, .bRequest = 0xA9, .wValue = 0x2344, .wIndex = 0x7F03, .wLength = 0 };
            static constexpr tusb_control_request_t INIT_3   = { .bmRequestType = 0x40, .bRequest = 0xA9, .wValue = 0x5839, .wIndex = 0x6832, .wLength = 0 };
            static constexpr tusb_control_request_t INIT_4   = { .bmRequestType = 0xC0, .bRequest = 0xA1, .wValue = 0x0000, .wIndex = 0xE416, .wLength = 2 };
            static constexpr tusb_control_request_t INIT_5   = { .bmRequestType = 0x40, .bRequest = 0xA1, .wValue = 0x0000, .wIndex = 0xE416, .wLength = 2 };
            static constexpr tusb_control_request_t INIT_6   = { .bmRequestType = 0xC0, .bRequest = 0xA1, .wValue = 0x0000, .wIndex = 0xE416, .wLength = 2 };
            //wValue in xbox360_wired_chatpad_command can be replaced with CHATPAD_CMD constants for different functions
            static constexpr tusb_control_request_t CMD         = { .bmRequestType = 0x41, .bRequest = 0x00, .wValue = 0x0000, .wIndex = 0x0002, .wLength = 0 };
            static constexpr tusb_control_request_t KEEPALIVE_1 = { .bmRequestType = 0x41, .bRequest = 0x00, .wValue = 0x001F, .wIndex = 0x0002, .wLength = 0 };
            static constexpr tusb_control_request_t KEEPALIVE_2 = { .bmRequestType = 0x41, .bRequest = 0x00, .wValue = 0x001E, .wIndex = 0x0002, .wLength = 0 };
            static constexpr tusb_control_request_t LEDS_1B     = { .bmRequestType = 0x41, .bRequest = 0x00, .wValue = 0x001B, .wIndex = 0x0002, .wLength = 0 };
        }
    }

    namespace Xbox360W
    {
        static constexpr uint8_t LED[] = { 0x00, 0x00, 0x08, 0x40 };
        //Sending 0x00, 0x00, 0x08, 0x00 will permanently disable rumble until you do this:
        static constexpr uint8_t RUMBLE_ENABLE[] = { 0x00, 0x00, 0x08, 0x01 };
        static constexpr uint8_t RUMBLE[]        = { 0x00, 0x01, 0x0F, 0xC0, 0x00, 0x00, 0x00 };
        static constexpr uint8_t INQUIRE_PRESENT[] = { 0x08, 0x00, 0x0F, 0xC0 };
        static constexpr uint8_t CONTROLLER_INFO[] = { 0x00, 0x00, 0x00, 0x40 };
        static constexpr uint8_t UNKNOWN[]      = { 0x00, 0x00, 0x02, 0x80 };
        static constexpr uint8_t POWER_OFF[]    = { 0x00, 0x00, 0x08, 0xC0 };

        namespace Chatpad 
        {
            static constexpr uint8_t INIT[] = { 0x00, 0x00, 0x0C, 0x1B };
            static constexpr uint8_t KEEPALIVE_1[] = { 0x00, 0x00, 0x0C, 0x1F };
            static constexpr uint8_t KEEPALIVE_2[] = { 0x00, 0x00, 0x0C, 0x1E };
            static constexpr uint8_t LEDS_ON_KEYPRESS[] = { 0x00, 0x00, 0x0C, 0x1B };

            static constexpr uint8_t CAPSLOCK = 0x20;
            static constexpr uint8_t GREEN = 0x08;
            static constexpr uint8_t ORANGE = 0x10;
            static constexpr uint8_t MESSENGER = 0x01;
            /*  The controller feedbacks the currently set leds. The bitmask for these are in chatpad_mod.
                xbox360w_chatpad_led_ctrl is used to turn on/off a chatpad led. Byte 3 is set to chatpad_led_on[x] 
                or chatpad_led_off[x] to turn that respective led on or off.    */
            static constexpr uint8_t LED_CTRL[] = { 0x00, 0x00, 0x0C, 0x00 };
            static constexpr uint8_t MOD[] =     { CAPSLOCK, GREEN, ORANGE, MESSENGER };
            static constexpr uint8_t LED_ON[] =  { 0x08, 0x09, 0x0A, 0x0B };
            static constexpr uint8_t LED_OFF[] = { 0x00, 0x01, 0x02, 0x03 };
        }
    }

```

Coolawy, 2026-01-29:
> Hi Guys, I have built a few diy 360 receivers recently with OGX.-Mini circuit in an rp2040zero module as a front end on Xbox original console. I have just discovered today the same as you that there is no motor rumble function.
> 
> After reading some of the posts here I have just gone back to retest the same controller and wireless hub but instead of wired opposites ogxmini hardware ryzee119s OGX360 (arduino based hardware **https://github.com/Ryzee119/ogx360** the motor rumble was just fine so it's not bricked my controllers.

Hi

I don't fully understand what you tried.
Did you use wireless adapter with OGX360 board connected to OG Xbox or did you use wire with wireless controller (change and play)? 

How can I try it without owning OG Xbox?
I have 4 controllers where rumble died and I would gladly fix them

faithvoid, 2026-01-31:
Good news, everyone! I spent the night throwing together a rumble fix script for Linux users in Python, the damage is **NOT** permanent! So far my vibration is back on my PC (the only system I have access to right now), but instantly shuts off the second it's connected to an OGX-Mini again. Someone with more time than me can cobble together a patch for OGX-Mini itself, I'm sure.

Not 100% sure if it works for wired controllers, but it's revived my wireless controller. Please give this a shot and let me know if you run into any issues!

https://github.com/faithvoid/OGXMiniRumbleFix

rebeloom, 2026-01-31:
> Good news, everyone! I spent the night throwing together a rumble fix script for Linux users in Python, the vibration issue is **NOT** permanent! So far my vibration is back on my PC (the only system I have access to right now), but instantly shuts off the second it's connected to an OGX-Mini again. Someone with more time than me can cobble together a patch for OGX-Mini itself, I'm sure.
> 
> Not 100% sure if it works for wired controllers, but it's revived my wireless controller. Please give this a shot and let me know if you run into any issues!
> 
> https://github.com/faithvoid/OGXMiniRumbleFix

a true hero! Recovered three controllers! thanks!

TRRAXT, 2026-02-24:
> Knowing that a total of 9 controllers are confirmed operational again warms my heart, this is what DIY's all about, helping each other out and making the best things we possibly can. :)

Hell yeah! Thank you [faithvoid](https://github.com/faithvoid)!
I'll test later on 2-3 my x360 controllers and give feedback!

Can someone find and fix (in fork?) some mistakes in ogx-mini that kills vibration?
So we can build it and be with controllers that have working vibration

Abro61209, 2026-02-24:
Thanks @alexcorvis84  

Is putting the ogx mini in webapp mode essentially the same as bootloader mode? Given you can install firmware through the website in webapp mode, I'd say yes?

faithvoid, 2026-02-24:
> ⚠️⚠️⚠️ For those who want to use X360 controller until solved the issue (it looks like it will take a while, as the project has not been updated for some time...) I'd recommend to try instead [Joypad-OS](https://github.com/joypad-ai/joypad-os). ⚠️⚠️⚠️
> 
> I can confirm it works like a charm and you don´t have to deal with the fear that the rumble will break down ;-)

Thanks for the software recommendation! Have you tried Joypad-OS on an OG Xbox, by any chance? Seems to work great PC-side, but none of the modes (including OG / XID mode, of course) work at all on my original Xbox (using an RP2350-USB-A). Closer to a better solution, but not quite there.

ashleylad, 2026-02-27:
@faithvoid.........this is fab news thanks for all your effort. I have a couple of the RP2350A modules will test in the morning. Are you able to update for the Rp2040 Zero module?

thanks

Ash

LeGittee, 2026-02-27:
> Amazing news, everyone! I got a compilation environment sorted and figured it out, I think!
> [...]
> [OGX-Mini-v1.0.0a3-RP2350_USB_A.uf2.zip](https://github.com/user-attachments/files/25611877/OGX-Mini-v1.0.0a3-RP2350_USB_A.uf2.zip)

Will this work on Xbox 360 in my waveshare RP2350 USB-A ? also does it have the XSM3 update ?

faithvoid, 2026-02-27:
> Will this work on Xbox 360 in my waveshare RP2350 USB-A ? also does it have the XSM3 update ?

The version I'm compiling is almost 1:1 from this fork, minus RP2350-specific changes, so it unfortunately won't work on unmodified Xbox 360 systems. Maybe @MegaCadeDev can implement support in his fork, as I don't have an Xbox 360 to test, and adding more features other than this fix is outside of my scope. My main goal here is just to publish a fix, it's up to other devs to implement it in their actively-maintained forks as I have way too many irons in the fire.

Abro61209, 2026-03-05:
Great news!

coldchick, 2026-03-22:
> > [@faithvoid](https://github.com/faithvoid).........this is fab news thanks for all your effort. I have a couple of the RP2350A modules will test in the morning. Are you able to update for the Rp2040 Zero module?
> > thanks
> > Ash
> 
> I'm genuinely fighting for my life against the compiler right now so I have no clue if this version works, but give it a shot. If anyone else needs a version compiled, please clone the repository yourself, make the required changes as stated above and follow the default repository build instructions, since my local version is a hodgepodge of RP2350-specific changes and it'll take me way longer to compile a non-RP2350 version than it will for anyone else to start fresh, and I don't intend on maintaining any specific fork, just contributing fixes where I can.
> 
> [OGX-Mini-v1.0.0a3-RP_ZERO.uf2.zip](https://github.com/user-attachments/files/25616535/OGX-Mini-v1.0.0a3-RP_ZERO.uf2.zip)

**Can you make firmware 1.0.0a3 for the chinese rp_zero boards?** 
Thanks a lot

[naething](https://github.com/naething)
[on May 31, 2025](https://github.com/wiredopposite/OGX-Mini/discussions/70#discussioncomment-13326578)
Comentario original en inglés - 

These clone boards aren't able to support 240 MHz reliably. They claim to be able to do 135 MHz. One big change between v0.4 and v1 alpha is the clock speed.

**I tried changing

#define SYSCLOCK_KHZ 240000

to

#define SYSCLOCK_KHZ 120000**

And then everything seems to work nicely. Not sure if changing the clock breaks anything else.