wiredopposite/OGX-Mini #149: Xbox Series Controller freezes (Pico W)

KierXYZ, 2025-06-16:
Seems to be having a repetitive issue over several original Xbox's while using an official Xbox One/ series controller on the Pico W Everything works correctly until booting a game. The controller stays connected but there is simply no input whatsoever. Once I reset the Pico power by unplugging and replugging, it works absolutely fine again. However, I have had it drop out in the same way again during gameplay. To note, I also use an official Pico W, no clones of any kind, and this is across 3 Pico W's and several original Xbox's over different revisions. Wondering if this is a bug or something that's already on the fix list for future updates. Haven't had this issue with PS4 controller in testing.

xmgprojects, 2025-11-02:
I had similar issues with an original Pico2 W on OG Xbox and Xbox controller v1914 connected wireless. I used the firmware from Team Resurgent repo and the issue has been resolved.

xmgprojects, 2026-02-02:
This is the one I'm using.
https://github.com/Team-Resurgent/OGX-Mini/releases

keving40-sudo, 2026-02-06:
> No problem. You flash the pico the same way you flased it originally. I would suggest to flash it first with nuke file to start clean. Select the appropriate fw version for your pico from Team Resurgent repo and drag and drop it into your pico.
> 
> Nuke file. Get the latest available. https://github.com/Gadgetoid/pico-universal-flash-nuke/tags

Preliminary testing (not an extended session by any means), show this worked. Thank you SO much for all your assistance.

sjftech, 2026-03-03:
The Team-Resurgent release seems to have the same issues for me where I have to unplug and re-plug the adapter to get the controller working once a game has launched, which is frustrating.