wiredopposite/OGX-Mini #177: reset required when using ogxmini with 8bit do dongle and wireless controllers

ashleylad, 2025-07-31:
Hi Guys, Running the latest Alpha on RPi2040zero on OG Xbox with 8bitdo dongle and a Xbox Series X (1914) controller. The controller pairs no problem but no response from sticks or buttons. If I press reset button on module everything works okay till reboot. Have tried all firmware versions on the 8bit do still the same . Any one else experienced this issue? thanks

xxcrashxx, 2025-09-02:
I'm having that issue with the one I bought from the etsy page on here while using an 8bitdo usb adapter 2 and a ps5 controller.  Oddly enough, I've also got a RPi2040zero I picked up from aliexpress and it's working perfectly after reboots.  I've made sure both the etsy ogx mini and the 8bitdo 2 are on the newest firmware.  Sorry this isn't much help, but hopefully it'll get some traction to get an answer.

ashleylad, 2025-09-02:
xxcrashxx thanks for that. It's been very frustrating. I have tried many different modules for the 2040 also the 2350 forked code and all revisions of firmware both for the zero and the 8bitdo dongle itself. I still have the same issue. 

Using any wired controller it's perfect even my cheap wired pc controller I got for £1 from a discount shop

This issue defo lies with the wireless bit. I also have the same issue if using an Xbox 360 rf receiver and controller. Controller syncs to receiver etc but no response from any input sticks or button until reset on the 2040 or 2350 module is pressed.

The only thing I have not tried is an official Rpi pico with a usb attached for the 8bitdo or 360rf module...that's my next move,

Thanks
Ash

Tuko-Mas, 2025-09-03:
I use an 8bitdo v2 on its firmware 1.5 and it works without any problem, as does the xbox 360 rf, whether the original, the aliexpress version or the one I made using a phat xbox 360 rf card and an rp2040

xxcrashxx, 2025-09-04:
This was the one I picked up. https://www.aliexpress.us/item/3256809294554391.html

Tuko-Mas, 2025-09-05:
@xxcrashxx That OGX-mini is very expensive, here in Mexico it costs 10 dollars.

ashleylad, 2025-09-16:
A little bit more testing last night. I managed to get my V1 brown 8bitdo dongle working from 2019...I updated the firmware to the latest...tested on rp2040 ogxmini and rp2350A and pico with usb port using an Xbox One 1708 controller. Bingo all worked just fine.

So retried black V2 dongle updated to the latest firmware and controller works fine after syncing no reset required!!! All works fine until rebooting then the 8bitdo dongle doesn't wake back up again and a resync is required every time
I have purchased 8bitdos from ebay amazon and ali...I wonder if they all  behave the same...the frustration continues but I have not give up...no yet!!

Tuko-Mas, 2025-09-17:
@ashleylad I use the 8bitdo v2 with firmware version 1.05, although the latest firmware works but the errors are not analogous. From what I understand, your problem is that you always have to synchronize it? The same thing happens to me, and well, at least it doesn't bother me.