wiredopposite/OGX-Mini #184: PS5/PS4 console

Kronos2308, 2025-08-15:
PS5 and PS4 consoles support

For example use a PS4 controller on a PS5 or any generic pad on a PS4 

I will be limited to the games that not required some special features like táctil pad or that make a simple use of that

mrdude2478, 2025-09-07:
> > PS4 has a controller 8 minute timout issue if not using a regular PS4 controller due to encyption/security keys needing sent in that time period, there's a Mayflash Magic-S Ultimate or Pro 2 that apparenty works on PS4 but not PS5. There's different versions of bluetooth - classic and LE so you need a device with the bluetooth version that has the same type of radio as the controller + device you are connecting to and is able to bypass the consoles security measures.
> 
> These guys did it https://gp2040-ce.info/

No they didn't you need a passthrough authentication device with PS4 encyption/decryption keys such as a mayflash or brook wingman.

Kronos2308, 2025-09-07:
@mrdude2478 nope 
I check the code and they do a proper auth with the PS4
There is a dump ``jedi_flash-Aug_3_2013.bin``
A PS4 controller firmware dump after reverse some of the code they get 3 files from the firmware 

   ``Private Key (PEM) - Choose your PEM file.
    Serial Number (16 Bytes in Hex Ascii) - Choose your serial number file.
    Signature (256 Bytes in Binary) - Choose your signature file.``

Key.pem,serial.txt,sing.bin

They are used to cypher the challenge of the ps4

Kronos2308, 2025-09-07:
BTW I made use of the latest release on a rp2040 Zero to connect a PS4 controller to a PS3 and the R3 in some point gets into turbo and start press R3 by itself 

WY use the device if PS3 has native support?
Well some Nacon controllers work on PS4 but not on ps3 with the rp2040 work and the PS button also work but the R3 start making some crazy stuff 

I tested this with a original PS4 and PS5 controller same R3 problem

Kronos2308, 2025-09-08:
Well you probably used the incorrect files
I connect my zero to the PS4 no USB passthrough device connected and after 20 minutes keeps going and can control the PS4
[Reddit post](https://www.reddit.com/r/fightsticks/comments/134sxo7/gp2040_now_supports_ps4_does_anyone_have_the_file/)

[image]

Kronos2308, 2025-09-08:
On the Zero pin 14 == PS button you have to ground that pin for second to press ps 

In mi case is 14 pin can be changed from web config

Kronos2308, 2025-09-08:
Yes, it's more or less like you say, except that I'm not using any control via USB, that firmware is oly support PS4 controllers so , yes you can just press ps on the controller but I was not usin

closed 2025-09-08 by Kronos2308

Kronos2308, 2025-09-08:
That firmware is made to use the pico as the controller by soldering buttons to the gpio pins
The USB is just a addon that suport DS4 controllers