wiredopposite/OGX-Mini #192: Compiling v1.0.0a4 branch for Pico W

JustMeDaFaq, 2025-08-27:
Hi,
got anyone the alpha 4 branch compiled for the Pico W?
Configuring it via:
`cmake -S . -B build -G Ninja -DCMAKE_BUILD_TYPE=Release -DOGXMINI_BOARD="PI_PICO_W" -DMAX_GAMEPADS=1 -DPICOTOOL_FETCH_FROM_GIT_PATH=1 -DPICO_SDK_FETCH_FROM_GIT=1`

yields in:
```
CMake Error at CMakeLists.txt:204 (target_include_directories):
  Cannot specify include directories for target "pico_btstack_classic" which
  is not built by this project.
```

I can be configured and build for the "non W" Pico without issues.

murilopereirame, 2025-10-23:
> This is actually because the latest commit uses submodule commits, that are not available online.
> 
> Basically wiredopposite pushed commits for ogx-mini, but not for the submodules it uses.
> 
> [@wiredopposite](https://github.com/wiredopposite) could you please push commits for the submodules too? Or alternatively release precompiled uf2. I want to test, whether the ps3-related issues were fixed.

Hey, I was able to compile a fork from OGX-Mini that apply some patches for Xbox 360 targeting the board PICO 2W, and it worked seamlessly, should work for the unpatched version too. So I'll leave my steps below:

**Fork:** https://github.com/barrenechea/OGX-Mini/tree/v1.0.0a4-patched
**Setup:** MacBook Air 13" M3 16GB with Docker 28.2.2, build e6534b4 running Ubuntu 24.04.3 LTS, Kernel 6.10.14-linuxkit

**Commands (in order):**

**Outside Docker Container:**
```bash
docker run -v ./data:/workspace -it ubuntu /bin/bash
```

**Inside Docker Container:**
```bash
cd /workspace

apt-get update && apt-get install -y \
    git \
    python3 \
    python3-pip \
    cmake \
    ninja-build \
    gcc-arm-none-eabi \
    libnewlib-arm-none-eabi \
    build-essential \
    pkg-config \
    curl \
    wget \
    ca-certificates
    
git clone https://github.com/barrenechea/OGX-Mini.git
git clone https://github.com/raspberrypi/pico-sdk.git

cd pico-sdk

git checkout 2.2.0
git submodule update --init --recursive
export PICO_SDK_PATH=$(pwd)

cd OGX-Mini

git checkout v1.0.0a4-patched
git submodule update --init --recursive

cd Firmware/pico

mkdir build

cd build

cmake .. -DCMAKE_BUILD_TYPE=Release -DOGXMINI_BOARD=PI_PICO_2_W -DMAX_GAMEPADS=1
make -j
```

It should produce a ```OGX-Mini.uf2``` inside the build folder, just copy and paste it inside the Raspberry Pi

throwaway29292, 2025-11-17:
> > This is actually because the latest commit uses submodule commits, that are not available online.
> > Basically wiredopposite pushed commits for ogx-mini, but not for the submodules it uses.
> > [@wiredopposite](https://github.com/wiredopposite) could you please push commits for the submodules too? Or alternatively release precompiled uf2. I want to test, whether the ps3-related issues were fixed.
> 
> Hey, I was able to compile a fork from OGX-Mini that apply some patches for Xbox 360 targeting the board PICO 2W, and it worked seamlessly, should work for the unpatched version too. So I'll leave my steps below:
> 
> **Fork:** https://github.com/barrenechea/OGX-Mini/tree/v1.0.0a4-patched **Setup:** MacBook Air 13" M3 16GB with Docker 28.2.2, build e6534b4 running Ubuntu 24.04.3 LTS, Kernel 6.10.14-linuxkit
> 
> **Commands (in order):**
> 
> **Outside Docker Container:**
> 
> docker run -v ./data:/workspace -it ubuntu /bin/bash
> 
> **Inside Docker Container:**
> 
> cd /workspace
> 
> apt-get update && apt-get install -y \
>     git \
>     python3 \
>     python3-pip \
>     cmake \
>     ninja-build \
>     gcc-arm-none-eabi \
>     libnewlib-arm-none-eabi \
>     build-essential \
>     pkg-config \
>     curl \
>     wget \
>     ca-certificates
>     
> git clone https://github.com/barrenechea/OGX-Mini.git
> git clone https://github.com/raspberrypi/pico-sdk.git
> 
> cd pico-sdk
> 
> git checkout 2.2.0
> git submodule update --init --recursive
> export PICO_SDK_PATH=$(pwd)
> 
> cd OGX-Mini
> 
> git checkout v1.0.0a4-patched
> git submodule update --init --recursive
> 
> cd Firmware/pico
> 
> mkdir build
> 
> cd build
> 
> cmake .. -DCMAKE_BUILD_TYPE=Release -DOGXMINI_BOARD=PI_PICO_2_W -DMAX_GAMEPADS=1
> make -j
> 
> It should produce a `OGX-Mini.uf2` inside the build folder, just copy and paste it inside the Raspberry Pi

I tried your branch on PI_PICO instead, but it just disconnects and never reconnects if i hold dpad down + start. I guess it's just broken on rp2040 then.

Pazureka, 2025-11-17:
I tried to compile normal, non patched 1.0.0.a4 version, but couldnt get it working, some compiling error. Xinput works just fine for me too. Theres just a problem with PS3 input, i think for some reason Switch mode (DPAD DOWN) works on PS3, but does not really support it, i figured it from a problem with PS button, it does not work, and left analog is inverted. When controller is connected in PS3 mode (DPAD Left), it does nothing.